sparkfun/nRF52832_Breakout #3: DFU Serial Instructions

boppana113, 2017-01-30:
Hi, I am using the Sparfun NRF52 breakout for a project and I want to develop my own PCB with it later. Is there any simple instruction guide on how to burn the bootloader program? 

Thank you.

jimblom, 2017-02-07:
I've cleaned up the Makefile, and added a bit more to the [bootloader README](https://github.com/sparkfun/nRF52832_Breakout/blob/master/README.md), which documents how to build the bootloader.

To burn the bootloader onto an nRF52832, you'll need a hardware programmer. I'd recommend grabbing an [nRF52832 Development Kit](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.nrf52/dita/nrf52/development/nrf52_dev_kit.html?cp=2_2_2), which can serve as a development platform for the nRF52 as well as a programmer for external chips.

Software tools like [nRFgo Studio](https://www.nordicsemi.com/eng/Products/2.4GHz-RF/nRFgo-Studio/) or [Nordic's command-line tools](http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.tools/dita/tools/nrf5x_command_line_tools/nrf5x_command_line_tools_lpage.html) can be used to load code onto the chip.

closed 2017-02-07 by jimblom

boppana113, 2017-03-31:
Hi Jimblom. I got the NRF52 DK and the Jlink EDU programmer. I was able to sucessfully program the sparkfun NRF52832 breakout board using Keil (J-link EDU). However, now I can't seem to go back and program through arduino. When I try to burn the bootloader using NRFgo studio and the bootloader hex file "sfe_nrf52832_dfu.hex" found in the repository it fails and gives an output saying:

[image]

Im not sure how to go around this. Thanks.

arihantdaga, 2018-08-03:
@boppana113 I am also stuck with this same thing for long time.. Did u find a solutions for this.. ?