sparkfun/nRF52832_Breakout #6: Flashing the bootloader

chaitzatkano, 2017-11-10:
Hi, 

Can you please confirm the command to flash the bootloader to sparkfun nrf52 breakout board?
I am trying to generate a zip of bootloader and softdevice using the following command but nrfutil throws error.

The command is 
nrfutil dfu genpkg bootloader.zip --bootloader sfe_nrf52832_dfu.hex --softdevice s132_nrf52_2.0.0_softdevice.hex  --dev-type 52 --sd-req 0x81

EliKrumholz, 2018-03-27:
Commenting in case it helps someone else. I needed to re-flash the bootloader on an nRF52 breakout board, I had to use a J-Link over the SWD connector, the nRF52 dev kit with onboard J-Link can be used: https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52-DK, but I used the J-Link Base. I needed to merge the provided hex files into a single hex file and then program the board using nrfjprog:

```
mergehex -m s132_nrf52_2.0.0_softdevice.hex sfe_nrf52832_dfu.hex -o merged.hex // Merge hex files
nrfjprog -e // Erase nRF52
nrfjprog -r --program merged.hex // Program the breakout board and reset
```

Afterwards the device entered bootloader mode as indicated by the blue LED blinking in the "timebomb" pattern and I was able to program the device over serial with an FTDI breakout board.

mhaberler, 2020-06-05:
MacOS Catalina user here - some more data points:

Acquired a [Black Magic Probe](https://1bitsquared.com/products/black-magic-probe) for gdb debugging

Managed to wipe the DFU bootloader with gdb 'monitor erase_mass' and figured I'll have to reflash it like [hinted here](https://learn.sparkfun.com/tutorials/nrf52832-breakout-board-hookup-guide?_ga=2.222170512.96291720.1591107048-157516391.1587736502#resources--going-further)

rebuilt bootloader from this repo as per [README](https://github.com/sparkfun/nRF52832_Breakout/tree/master/Firmware/bootloader-custom) using the [9-2020-q2-update armgcc](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) - the MacOS package installs in /Applications/ARM so Makefile.posix needs adjusting for this location

eventually figured out this build was not necessary in the first place since the [hex files](https://github.com/sparkfun/nRF52832_Breakout/tree/master/Firmware/bootloader-custom/hex) are just fine
merged those hex files as described above:
````
mergehex -m s132_nrf52_2.0.0_softdevice.hex sfe_nrf52832_dfu.hex -o merged.hex

````
attached the BMP SWD wires and supplied power from the serial black and red wires to GND and VIN (probably pin 3V3 works as well as power from the BMP is 3.3V)

loaded the bootloader like so:

````
$ /Applications/ARM/bin/arm-none-eabi-gdb
target extended-remote /dev/cu.usbmodemC1DDB5BD1
# turns on power on the serial VCC wire
monitor tpwr enable 
# check what's out there
monitor swdp_scan
att 1
# wipe everyting
monitor erase_mass
# load the merged bootloader
file <path to>/merged.hex
# flash it
load
# verify this worked
compare-sections
# disconnect
kill
# confirm disconnect
# blue LED should show 'timebomb pattern'
````

I assumed I would be able to serially program the board with the BMP serial port - for reasons which I do not understand this does not work

however, connecting a normal FTDI USB serial converter to GND/TX/RX and powering the board as before from the BMP does work, also from Arduino

I did have issues with the FTDI adapter as well until I installed the [2.4.4 FTDI MacOS driver](http://www.ftdicommunity.com/index.php?topic=505.0) - all the 'FTDI serial dongles do not reappear after disconnect/reconnect' issues are gone

still interested in getting the BMP serial port to work but at least back to a working config